OpenBoardView/OpenBoardView #285: pdf  search

moussatohme, 2023-04-27:
hello im having issue using pdf search button, when i click it nothiing happens, pdf already opened, im using mac.

piernov, 2023-04-28:
There is no support for PDF software integration on macOS.
It is only available on Windows with a custom build of SumatraPDF and Linux with a custom build of Evince.

I will have to hide the button to avoid the confusion.

yaourtiere, 2024-05-29:
This is certainly details that should be specified, at least in the Readme.
I've seen other people having trouble with this functionnality, it's a good point to hide it if not available, but the specific setup could be precised.

For example, I've searched for the name of this «specific» build, and cannot find it anywhere. There is <https://github.com/OpenBoardView/OpenBoardView/issues/296>, where it seems to be it but nowhere else.

Yes there is an IRC channel, and yes it is certainly experimental, but this should be specified clearly. I mean it is impossible to miss the "PDF Search" big button on the interface.

piernov, 2024-05-29:
@yaourtiere 
You can send pull requests to improve the features or the README.

The custom build of SumatraPDF for Windows is included in the Windows archive ( https://github.com/OpenBoardView/OpenBoardView/releases/download/9.95.0/openboardview-9.95.0-win32.zip ), the custom build of Evince is provided as a Flatpak package next to the other downloads ( https://github.com/OpenBoardView/OpenBoardView/releases/download/9.95.0/evince-43.alpha12707f9.flatpak ).
The source code repositories for these custom builds are available at https://github.com/OpenBoardView/sumatrapdf and https://github.com/OpenBoardView/evince .
As indicated, there is no PDF reader that has been adapted to work with OBV on macOS, so this is the specific situation where the button should not show up.

All of this is still part of a Beta release of OBV (9.95.0, added starting with 9.90.0), although I removed the pre-release tag so that people find it more easily because it should be preferred over the latest "stable" release anyway since it brings in a bunch of bug fixes and support for GenCAD file format as well.

There is no reason to claim it is a stable release until the PDF support is more user-friendly, which would require built-in support in SumatraPDF and Evince without extra installation steps or configuration needed. Right now, if you use either of these custom builds on a system that's already running SumatraPDF/Evince, it will most likely not work as intended due to conflicts in the IPC (Inter-Process Communication) mechanisms.

For Evince, my merge request to include the required additions upstream has been open for 2 years. https://gitlab.gnome.org/GNOME/evince/-/merge_requests/465
For SumatraPDF, I didn't make a pull request because I wasn't satisfied with how it would work yet. Especially the fact that, if it was included in the upstream release, it would still require manual configuration specifically for OBV since the path to the binary to call for inverse search has to be specified in SumatraPDF's configuration (in the custom build, it is set by default https://github.com/OpenBoardView/sumatrapdf/commit/5fe1deab66033af88f8ea4acfffa80858eb1cf81 ).

Support for macOS would require adapting yet another open-source PDF reader such as Skim to provide the couple of features needed for bi-directional search. On an operating system I'm not comfortable with, in a programming language I'm not familiar with, and with IPC mechanisms I know nothing about, that's why back when the experimental PDF support was added, this was not considered.

Again, contributions are welcome.