OpenBoardView/OpenBoardView #289: Multi-ASC files only supported in version 9.0.2 and not supported in later versions (Windows)

barzottotec, 2023-05-19:
[ASUS PRIME B250M-A{BOARDVIEW}.zip](https://github.com/OpenBoardView/OpenBoardView/files/11518382/ASUS.PRIME.B250M-A.BOARDVIEW.zip)

Hello. I'm using Windows and trying to open the file "ASUS PRIME B250M-A{BOARDVIEW}" attached, I have the FZ keys, tried versions 9.95.0, 9.90.0, 9.0.3, the error message is: 
There was an error opening the file: < path >
/home/piernov/devel/OpenBoardView/src/openboardview/FileFormats/ASCFile.cpp:60: ASCFile::parse_pin(char*&, char*&, char*&, char*&, line_iterator_t&)::<lambda()>: Assertion `value >= 0' failed.

[image]

Only version 9.0.2 opens the file normally. 

By the way, it's not possible to copy the error message, I had to write it manually here.

barzottotec, 2023-05-19:
> Lines 6168-6175 and 6250-6257 of PINS.ASC , pins 59 to 66 of JM21 and JM22
> all have missing properties. Not sure if it can only happen for this
> specific case. Would be quite tricky to handle in a general case (i.e. any
> missing property for any item).
> Anyway, 20c971d
> on master already partially fixes this specific case as a side-effect but
> it'll still render unconnected pins at (0,0).

I tried other files, so far this is the only one showing the error. Thanks
for the reply.

inflex, 2023-05-20:
In the ASCFile code where we parse the pin, we can resolve this issue by testing to see if p is not at the EOL.

The hiccup though will be that you'll need to read in the pin ID/number/name explicitly else the pin numbers will slip for each time we return early.

```
		pin.part = parts.size();
		pin.snum = READ_STR(); 
		pin.name = READ_STR();

		if ((*line_it) -p < 2) return;      // Test to see if p is too close to the EOL, if so, just move to next line.

		double posx = READ_DOUBLE();
		pin.pos.x   = posx * 1000.0f;
		double posy = READ_DOUBLE();
```