OpenBoardView/OpenBoardView #290: Part info like in BoardViewer - is it possible?

laszlo-pl, 2023-05-20:
Hi,

Is it possible to display details of a selected element in OBV as I can in BoardViewer?

If not, is there a plan to include this functionality in future OBV releases?

Regards
laszlo-pl

[image]

inflex, 2023-05-21:
That information is file-dependent (I'm presuming it's not a BRD file).

Do you have the example file ?

There are plans to have OpenBoardDATA support added to OBV which will provide this type of metadata too.

laszlo-pl, 2023-05-21:
Thank you for your answer.

In the attached image you can see a comparison between BV and OBV. Specifically, what information about the element are they showing me.

I also attach the files that I used for this comparison.
[ASUS P5GC-VM.zip](https://github.com/OpenBoardView/OpenBoardView/files/11523867/ASUS.P5GC-VM.zip)
[image]

inflex, 2023-05-21:
Thanks for that. In this case, yes, the data is contained within the parts.asc or BOM file as opposed to being available on the OBData site.   That said, most likely when the OBData extension is written this will be something that can be incorporated too.

Thanks for the sample file,  that helps a lot.

laszlo-pl, 2023-05-21:
I am glad that I could help and look forward to further development of the program.

closed 2023-05-21 by laszlo-pl