OpenBoardView/OpenBoardView #291: Teboview (tvw) format support request

swiftgeek, 2023-05-29:
Noticed that there isn't an issue about it yet in this repo.

* https://www.eevblog.com/forum/eda/wanted-teboview-tvw-file-format-specification-to-write-openboardview-decoder/
* https://github.com/inflex/teboviewformat

inflex, 2023-05-29:
I've been working on it incrementally.  Thankful for tools like ImHex being out there that have helped parse through the binary format.

Unfortunately due to other constraints in life I've sort of hit a wall for now.

swiftgeek, 2023-05-29:
I wonder if there is something that could make it easier, like generating `*.tvw` from particular/smaller layout (not that I could do that, but maybe there is somebody out there)

inflex, 2023-05-29:
Have done that; some very nice minimal units here - combined that with having the official teboview decoder in a windows VM so various individual bits/bytes could be changed to see their correlations.

Got to about this point with the decoding

[image]

qwertychouskie, 2024-04-11:
> Have done that; some very nice minimal units here - combined that with having the official teboview decoder in a windows VM so various individual bits/bytes could be changed to see their correlations.
> 
> Got to about this point with the decoding
> 
> ![Screenshot at 2023-05-29 23-48-07](https://private-user-images.githubusercontent.com/5660591/241745674-9a422c76-3919-401e-8afb-3c7ef261fd27.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTI4NzU5OTIsIm5iZiI6MTcxMjg3NTY5MiwicGF0aCI6Ii81NjYwNTkxLzI0MTc0NTY3NC05YTQyMmM3Ni0zOTE5LTQwMWUtOGFmYi0zYzdlZjI2MWZkMjcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDQxMSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA0MTFUMjI0ODEyWiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9N2JkM2JlZjJjMjI3Yjg2MjFiZjkyZjZmZjc4NjkzOWI3MmNmZmMzMzVmNzkzMGFjYjViODFhNDZlYjhjMGRhNCZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.bmt-9wqmn2y_WpXl7iOy-8fbRpTujC30rXBwh1tw7mA)

I saw on https://www.badcaps.net/forum/troubleshooting-hardware-devices-and-electronics-theory/troubleshooting-laptops-tablets-and-mobile-devices/schematic-requests-only/88901-wanted-teboview-tvw-file-format-specification-to-write-openboardview-decoder?p=3155564#post3155564 that you got much further, any update on when support will be merged?

inflex, 2024-04-12:
@qwertychouskie 
>> any update on when support will be merged?

Not for a while yet, that was in FlexBV, not OBV.  

I did have a bounty out for TVW decoder which would have been open sourced but no one claimed the bounty so ultimately after 2 years I threw my own resources in to it and now I need to recoup the costs.

I would say 9~12 months before I release the spec and someone else can implement it for OBV.

githubisnonfree, 2024-10-15:
Is the file named `fileformat-tvw.txt` in your repository <https://github.com/inflex/teboviewformat> sufficient as a reference for implementation? I also saw your video which referenced it: https://www.youtube.com/watch?v=tv1bziPegO0

This, in addition to your previous video calling for support. I'd have a crack at this myself, since I also need this support to be present in OBV. If this reference linked above is incomplete, and there is more information pertaining to TBV format, could you release what you know?

(short of implementing it in OBV of course, which would be my preference, otherwise I'd tempted to implement it myself based on this spec)

I was looking at a file today in that format. I understand you want to sell commercial licenses with your proprietary fork of OBV and that's your choice, but my interest is in openboardview exclusively, since I really only want to use free software (price isn't the issue either, your prices are reasonable, but I'm opposed to running proprietary stuff needlessly!)

inflex, 2024-10-15:
> I was looking at a file today in that format. I understand you want to sell commercial licenses with your proprietary fork of OBV and that's your choice, but my interest is in openboardview [...]

The data provides at the github link page isn't sufficient for decoding the format as yet; there's some painful obscure inferred indexing that makes you rip your hair out for a while.

Eventually I will release a spec for the format, most likely after I have done more work on it to make it more reliable and also decoding other aspects of the data.

cyrozap, 2026-01-09:
@inflex Any updates on releasing a spec for the TeboView format or merging support for it into OBV?

FearlessArmada, 2026-08-16:
I have implemented TVW almost completely now with references in this thread as well as reverse engineering other viewers. Layer viewing works okay-ish with some graphical quirks to work out. I am currently working on getting the same to work for XZZ formats at which point I will most likely create a PR.