OpenBoardView/OpenBoardView #292: File .FZ

juf00th, 2023-06-25:
Is support for ".FZ" files planned? If so, when?

inflex, 2023-06-25:
Been supported right from the start almost.

You will need the key to unlock the FZ files though, something we can't povide here but should be readily available with a google.

juf00th, 2023-06-25:
> Been supported right from the start almost.
> You will need the key to unlock the FZ files though, something we can't povide here but should be readily available with a google.

Yes, I googled some code and added it to the settings. And it works. Thank you. I understand this code can be different?

inflex, 2023-06-25:
Sometimes there are FZ files distributed that aren't encoded, that's about the only alternative version I've found.

closed 2023-06-25 by juf00th