OpenBoardView/OpenBoardView #296: PDF Search on Linux

nullibyte, 2023-08-02:
I'm trying to install evince-43.alpha12707f9.flatpak to try out pdf search, however I get the following error:

error: The application org.gnome.Evince/x86_64/master requires the runtime org.gnome.Platform/x86_64/master which was not found

I've tried installing runtime/org.gnome.Platform/x86_64/44 but this does not seem to help

marcoms75, 2023-11-13:
I installed it from the terminal on Ubuntu:

sudo apt install flatpak
flatpak remote-add --if-not-exists flathub https://flathub.org/repo/flathub.flatpakrepo
flatpak remote-add --if-not-exists flathub-beta https://flathub.org/beta-repo/flathub-beta.flatpakrepo
sudo flatpak install evince-43.alpha12707f9.flatpak

The problem is that evince-43.alpha12707f9 is installed but it doesn't start

piernov, 2025-08-20:
New Flatpak of the custom Evince build based on 48.1 has been released alongside OpenBoardView 9.95.2. It depends on org.gnome.Platform 48 and dependencies should be pulled automatically from Flathub.

closed 2025-08-20 by piernov