OpenBoardView/OpenBoardView #298: Problem parsing $SHAPES section when opening a GenCAD file

cyrozap, 2024-02-17:
I get the following error when I try to open a certain `.cad` file:

> Failed to parse GenCAD file: the $SHAPES section was not parsed properly

The file in question can be found [here](https://drive.google.com/file/d/1TB6mSRDPAEsk_usSitPlUJI31NzfqX8E/view) ([alternate link](https://drive.google.com/file/d/1_ycwGH52NJwSvoa25eSjcxI8JDGHtMh1/view)).

OpenBoardView version is latest from git (8528b80).

Elextrixman, 2024-02-17:
Try this. Tested working.

[TUF GAMING B650-PLUS WIFI 1.04.cad.zip](https://github.com/OpenBoardView/OpenBoardView/files/14319805/TUF.GAMING.B650-PLUS.WIFI.1.04.cad.zip)

Elextrixman, 2024-02-17:
Used 2.0.1.9 Beta. And it opened ok.

Elextrixman, 2024-02-17:
Drag and drop in here
[BoardViewer.zip](https://github.com/OpenBoardView/OpenBoardView/files/14319823/BoardViewer.zip)

swiftgeek, 2024-02-18:
That file is crashing CAMCAD (4.14.312.6), so something went horribly wrong there

piernov, 2024-02-26:
Similar issue with this one where the `$SHAPES` section does not start with `SHAPE` but instead `INSERT`. Not sure what to do here, seems like a bogus file but if there's a tool that actually produces these maybe we should try to take it into account (but that's a pain).
[Asus_X1403ZA_BV.zip](https://github.com/OpenBoardView/OpenBoardView/files/14410392/Asus_X1403ZA_BV.zip)

swiftgeek, 2024-02-27:
I think a conversion/repair script would be more feasible here, since those files are so deeply broken (and thus most of the contents those files can be discarded since it cannot be correlated back)

piernov, 2024-07-24:
Workaround from 46c603c should work here as well. The original file is still a bit weird so let's hope it renders mostly fine.

closed 2024-07-24 by piernov