OpenBoardView/OpenBoardView #302: Some parts with asymmetrical pins do not get enclosed in a rectangle

sabogalc, 2024-04-29:
I have a boardview file with a few components where the pins are not symmetrical (e.g. one side of the component has three pins and the other side only has one).

If you look at J9, this can be seen with 2 pins on one side vs. 1 pin on the other, but the component is properly enclosed in a rectangle. On U1 however, the component shape is a bit weird.
[image]
This is what the same boardview file looks like in FlexBV.
[image]
I've also attached the boardview file with a `.txt` extension so that it could be uploaded to this ticket. The original extension was `.bvr`, but I was able to open it with OpenBoardView in the `.txt` format just fine.
[Arduino UNO.txt](https://github.com/OpenBoardView/OpenBoardView/files/15144807/Arduino.UNO.txt)

inflex, 2024-04-29:
The outline is generated by heuristics "rectangled best fit" type algorithm so sometimes the results vary.  You can fix it for one instance of a pin arrangement but then it throws out others.   

Unfortunately without part outline data it's always a guess.

sabogalc, 2024-04-29:
Does this also apply to FlexBV, and if so did its algorithm get the shape correct just because its method and successes/failures are different?

I'm not a programmer so forgive me if I'm oversimplifying the issue, but in the OpenBoardView screenshot it looks like for U1 and J9 the rectangle is formed by aligning one of its sides to one of the internal straight edges of the triangle. Could this problem be solved by having that rectangle edge pair with the triangle edge that has the most pins on it?

inflex, 2024-04-29:
FlexBV uses a different algorithm, so yes, the results are different. 

Essentially what OBV/FBV try to do is draw a just-fits quad (rectangle) around the pins at varying angles until it finds the one with the minimum area and subsequently selects that one.

Most of the time it works correctly, now and then you do get incorrect results, however, trying to improve the outcome of a specific part ( ie, U1 in this case ) often results in other parts on other boards being displayed incorrectly.

In summary, it's the "best of an imperfect set of options" as opposed to being a bug per`se.

Also, latest versions of OBV should support BVR (  https://github.com/OpenBoardView/OpenBoardView/pull/236 )