OpenBoardView/OpenBoardView #305: GenCad "XY html to CAD V1.2" support

moldimolt, 2024-07-22:
Hello, 

OpenBoardView cannot view the attached GenCad 1.4 file, reporting it is invalid. There was a commit merged in 2023 to support these files but it seems to be broken for this file specifically.

Thank you.

[GX650PI 2.0 60NR0D70-MB1020 GX650PI MB._0M_R9-7945HX__R2.0(V8G_SAMSUNG)_PERKEY_TP_PD_AERO.zip](https://github.com/user-attachments/files/16327234/GX650PI.2.0.60NR0D70-MB1020.GX650PI.MB._0M_R9-7945HX__R2.0.V8G_SAMSUNG._PERKEY_TP_PD_AERO.zip)

piernov, 2024-07-22:
It does take a very long time to load (almost 2 minutes on my machine), I don't know why, but once it's loaded it looks ok.
OpenBoardView built from the last commit on master branch.
[image]

moldimolt, 2024-07-22:
Thanks! I took another look and it seems the issue is the commit adding support was in June 2023, but there hasn't been beta or stable release since Feb 2023. I will try compiling from the latest commit.

piernov, 2024-07-22:
Indeed, I will make another pre-release to include the bunch of GenCAD fixes and workarounds from the past year, hopefully soon.

piernov, 2024-07-24:
Here it is: https://github.com/OpenBoardView/OpenBoardView/releases/tag/9.95.1

closed 2024-07-24 by piernov

mumu170509, 2024-09-10:
where can i download  XY html to CAD V1.2 utility
thankyou 
@moldimolt 
> Hello, 
> 
> OpenBoardView cannot view the attached GenCad 1.4 file, reporting it is invalid. There was a commit merged in 2023 to support these files but it seems to be broken for this file specifically on the latest stable release.
> 
> Thank you.
> 
> [GX650PI 2.0 60NR0D70-MB1020 GX650PI MB._0M_R9-7945HX__R2.0(V8G_SAMSUNG)_PERKEY_TP_PD_AERO.zip](https://github.com/user-attachments/files/16327234/GX650PI.2.0.60NR0D70-MB1020.GX650PI.MB._0M_R9-7945HX__R2.0.V8G_SAMSUNG._PERKEY_TP_PD_AERO.zip)
>