OpenBoardView/OpenBoardView #310: [MacOS] Failed on Installation of OpenBoardViewer

quynhcvn, 2024-12-23:
I got the following error when install the OpenBoardView. This happened with I ran the command ./build.sh. Could you please help to review and advise?
(I used MacOS Sequoia 15.2)

```
c++: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [src/openboardview/openboardview.app/Contents/MacOS/openboardview] Error 1
make[1]: *** [src/openboardview/CMakeFiles/openboardview.dir/all] Error 2
make: *** [all] Error 2
MAKE INSTALL/STRIP FAILED
```
This is details of logs when running ./build.sh
[OpenBoardView installation.txt](https://github.com/user-attachments/files/18225100/OpenBoardView.installation.txt)

OllieKrause, 2025-03-12:
@quynhcvn I just had the same issue. Your linking error is likely because you're running on apple silicon (this part of your error message was truncated in your initial post though). Your system is compiling for x86_64 but your SDL2 library is for ARM64. I solved this by reinstalling a x86_64 version of homebrew and then reinstalling the corresponding sdl2.

1. install x86_64 Homebrew
`arch -x86_64 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"`

2. Install SDL2 for x86_64 Homebrew
`arch -x86_64 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/HEAD/install.sh)"`

3. Try building openboardview
`export CMAKE_OSX_ARCHITECTURES="x86_64"`
`rm -rf release_build`
`arch -x86_64 ./build.sh`

piernov, 2025-03-12:
Use the official SDL2 Framework build, not homebrew.
e.g., https://github.com/libsdl-org/SDL/releases/download/release-2.32.2/SDL2-2.32.2.dmg