OpenBoardView/OpenBoardView #312: BRD file not recognized (MacOS)

loopless, 2025-02-06:
I had no problem building and running under MacOS ( great job!). Except I did need to remove the x86_64 flag to CMake.

I am trying to open a .BRD file shipped with Eagle 9.3.0 and it fails with "unrecognized file format"

[hexapod.brd.zip](https://github.com/user-attachments/files/18697028/hexapod.brd.zip)

sabogalc, 2025-02-12:
This is not a bug with the software, but rather with the `.brd` file.

EAGLE uses `.brd` as an extension for their PCB files, and these are not boardview files.

As an example, below is a photo showing the Arduino UNO R3 EAGLE `.brd` file in a plaintext editor (obtained from the CAD Files section [here](https://docs.arduino.cc/hardware/uno-rev3/)).

[image]

Now, below is a photo of the text for a boardview `.brd` file for that same board.

[image]

Likewise, when I try to open the EAGLE `.brd` file, OpenBoardView will tell me that it is not compatible.

[image]

However, opening the boardview `.brd` file works.

[image]

Looking at your attached file, it matches the EAGLE format and not the boardview format, and that is why the software cannot read it. To help you out, I did my best to convert the board into a boardview file, which you can see below and I've attached it as a `.zip`.

[image]

[hexapod.brd.zip](https://github.com/user-attachments/files/18761985/hexapod.brd.zip)

loopless, 2025-02-12:
Thanks for the detailed explanation and debugging of the file. This was an Eagle sample file I was using to experiment with the software.

sabogalc, 2025-02-12:
Please let me know if you would like any other information or file conversions, and if not then feel free to close the issue.