OpenBoardView/OpenBoardView #327: GCC 15 in Arch Linux repos cause a SIGSEGV on launch after compiling from source

FearlessArmada, 2025-05-08:
I spent probably two hours trying to figure out what my issue was when I manually set CMake to use GCC-13 instead of the default GCC-15 and it runs just fine.

The error was the following:

`Process finished with exit code 139 (interrupted by signal 11:SIGSEGV)`

Debugging in Clion showed the following:

[image]

GCC that makes faulty executable and GCC that makes the working executable version outputs:

```Es werden eingebaute Spezifikationen verwendet.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/15.1.1/lto-wrapper
Ziel: x86_64-pc-linux-gnu
Konfiguriert mit: /build/gcc/src/gcc/configure --enable-languages=ada,c,c++,d,fortran,go,lto,m2,objc,obj-c++,rust,cobol --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://gitlab.archlinux.org/archlinux/packaging/packages/gcc/-/issues --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --enable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror
Thread-Modell: posix
Unterstützte LTO-Kompressionsalgorithmen: zlib zstd
gcc-Version 15.1.1 20250425 (GCC)

--------------------------------------------

Es werden eingebaute Spezifikationen verwendet.
COLLECT_GCC=gcc-13
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-pc-linux-gnu/13.3.1/lto-wrapper
Ziel: x86_64-pc-linux-gnu
Konfiguriert mit: /build/gcc13/src/gcc/configure --enable-languages=c,c++,fortran --enable-bootstrap --prefix=/usr --libdir=/usr/lib --libexecdir=/usr/lib --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=https://gitlab.archlinux.org/archlinux/packaging/packages/gcc13/-/issues --with-build-config=bootstrap-lto --with-linker-hash-style=gnu --with-system-zlib --enable-__cxa_atexit --enable-cet=auto --enable-checking=release --enable-clocale=gnu --enable-default-pie --enable-default-ssp --enable-gnu-indirect-function --enable-gnu-unique-object --enable-libstdcxx-backtrace --enable-link-serialization=1 --enable-linker-build-id --enable-lto --disable-multilib --enable-plugin --enable-shared --enable-threads=posix --disable-libssp --disable-libstdcxx-pch --disable-werror --program-suffix=-13 --enable-version-specific-runtime-libs
Thread-Modell: posix
Unterstützte LTO-Kompressionsalgorithmen: zlib zstd
gcc-Version 13.3.1 20241125 (GCC)

FearlessArmada, 2025-05-09:
> Try with the changes from [#320](https://github.com/OpenBoardView/OpenBoardView/pull/320) .

That worked, interesting because I thought it was a patch for Windows but seems not.

troed, 2025-06-25:
Adding that the #320 PR solved the same issue for me just now. GCC 15 on Asahi Fedora (aarch64).

closed 2025-08-11 by piernov