OpenBoardView/OpenBoardView #329: Build Success on M4 Mac (Apple Silicon) with build.sh and Codesign Fixes

iwskd, 2025-06-23:
**Title: Build Success on M4 Mac (Apple Silicon) with `build.sh` and Codesign Fixes**

Hello OpenBoardView Team and Community,

I'm writing to report a successful build and execution of OpenBoardView on a new Apple Silicon Mac mini with an M4 chip, along with the specific steps and a minor modification needed to the `build.sh` script to achieve this. This information might be helpful for other users with similar hardware or for future updates to the build process.

---

### My System Specifications:

* **Model Name:** Mac mini
* **Model Identifier:** Mac16,10
* **Model Number:** MU9D3J/A
* **Chip:** Apple M4
* **Total Number of Cores:** 10 (4 performance and 6 efficiency)
* **Memory:** 16 GB
* **System Version:** macOS 15.5 (24F74)

---

### Problem Encountered (Initial Build Failure):

After cloning the repository and running `./build.sh` as per the README, the build process failed during the linking stage with `Undefined symbols for architecture x86_64` errors. This indicated that while I was on an `arm64` system, the build process was attempting to create a `x86_64` (or Universal 2) binary, leading to a mismatch with the `arm64` libraries installed via Homebrew (e.g., `libSDL2.dylib`).

A relevant line from my initial `build.sh` output showing the architectures being targeted was:
Extra flags passed to CMake: -DCMAKE_INSTALL_PREFIX= -DCMAKE_OSX_ARCHITECTURES=arm64;x86_64

---

### Solution (Modification to `build.sh` and Codesigning):

To resolve this, I made a minor but crucial modification to the `build.sh` script:

1.  **Modified `build.sh`:**
    I opened `build.sh` and changed the line that sets `CMAKE_OSX_ARCHITECTURES`.

    **Original line:**
    ```bash
    if [[ "$(uname -a)" == *"arm64"* ]]; then
      COMPILEFLAGS="$COMPILEFLAGS -DCMAKE_OSX_ARCHITECTURES=arm64;x86_64"
    fi
    ```

    **Modified line (to target `arm64` exclusively):**
    ```bash
    if [[ "$(uname -a)" == *"arm64"* ]]; then
      COMPILEFLAGS="$COMPILEFLAGS -DCMAKE_OSX_ARCHITECTURES=arm64"
    fi
    ```
    This ensures that on Apple Silicon Macs, only the `arm64` architecture is targeted, preventing conflicts with `x86_64` dependencies.

2.  **Re-ran `build.sh`:**
    After this modification, `./build.sh` completed successfully in terms of compilation and linking. The output confirmed `Extra flags passed to CMake: -DCMAKE_INSTALL_PREFIX= -DCMAKE_OSX_ARCHITECTURES=arm64`.

3.  **Addressed Code Signature Invalidity:**
    Upon attempting to run the newly built `openboardview.app`, it crashed immediately with a `SIGKILL (Code Signature Invalid)` error. This is typical for self-built applications on modern macOS versions due to System Integrity Protection (SIP) and Gatekeeper.

    To fix this, I applied an ad-hoc code signature to the application:
    ```bash
    codesign --force --deep --sign - ./openboardview.app
    ```
    The output `replacing existing signature` indicated success.

---

### Result:

After performing these steps, `openboardview.app` launched successfully and appears to be fully functional on my M4 Mac mini.

Thank you for providing such a valuable open-source tool! I hope this report helps in refining the build process for Apple Silicon devices, especially newer chips like the M4.

Best regards,

Takuya Iwasaki / iwskd

inflex, 2025-06-23:
This is very good information and thank you for posting it.   Likely it'll be useful for a lot more than just the OBV community too as I imagine a number of projects are wondering how to compile properly on the M-series machines.  

Thank you.

piernov, 2025-06-23:
As written in #310 , you need to use the official SDL2 Framework that contains both the x86_64 and arm64 architectures in order to build a universal executable targeting both, not Homebrew library which contains only the host architecture.
The codesign command is in the README.md.

galkinvv, 2025-07-03:
Actually, OpenBoardView has GitHub actions enabled that are building artefacts (stored for 3 months since last commit and downloadable only by logged-in users, those are GitHub limitations)

In theory, the universal build compatible with M4 is generated as _openboardview_unsigned_macos15_universal_dmg.COMMIT_ID_ but unfortunately:

- not sure if it was ever tested upon actions were introduced
- until #330 it lacked sign completely, even the ad-hoc one

@iwskd if you are interested in getting OpenBoardView releases for M4 - can you please test the openboardview_unsigned_macos15_universal_dmg build artifacts from [adhoc-sign-enabled-PR](https://github.com/OpenBoardView/OpenBoardView/actions/runs/16038976910?pr=330) and the [earlier from master branch](https://github.com/OpenBoardView/OpenBoardView/actions/runs/14724430154)

The main questions are:

- Are those builds usable on your M4 machine? After whitelisting the app in Privacy & Security settings, I suppose?
- Is their behaviour identical, or the newer "adhoc-sign-enabled-PR build" behaves better?

From those questions, I want to understand if the PR enabling ad-hoc signing in GitHub actions make sense or behaviour of those builds is equally good/bad.

jlafitte, 2025-10-20:
This worked for me, thanks.

galkinvv, 2025-10-20:
@jlafitte since you seems to be using Openboadview on mac - can you provide some feedback regarding .dmgs produced by github actions?

Curretly on master we have code signing disabled, and there is a PR brancg that enables ad-hoc signing. But I'm not sure if this is actually an improvement (and no envorinment to test it on modern macs).

Can you compare the 
`openboardview_unsigned_macos15_universal_....dmg` build artifacts from [adhoc-sign-enabled-PR](https://github.com/OpenBoardView/OpenBoardView/actions/runs/18650245272) and the [earlier from master branch](https://github.com/OpenBoardView/OpenBoardView/actions/runs/18230998787) ?

The main questions are:

* Are those builds usable on your M4 machine? After whitelisting the app in Privacy & Security settings, I suppose?
* Is their behaviour identical, or the newer "adhoc-sign-enabled-PR build" behaves better?